tier4/nebula #52: Gap in the point cloud on Velodyne sensor

mebasoglu, 2023-08-15:
## Description
When I use Nebula with `VLP-16`, I saw a gap on the point cloud. The gap is flickering. Here is a video of the point cloud:

https://github.com/tier4/nebula/assets/13589149/ba3d4cce-0661-409f-b08e-e2f69d7aceb1

## Expected behavior
We should have a full point cloud without any gaps.

## Actual behavior
There is gap which is flickering on the point cloud.

## Steps to reproduce
Run the launch file for Velodyne on the main branch:
```bash
ros2 launch nebula_ros velodyne_launch_all_hw.xml sensor_model:=VLP16 sensor_ip:=192.168.1.208 data_port:=2368
```
And on the RViz2 screen the point cloud:
[image]

## Additional context
I can provide a pcap file that I recorded from the sensor if it is needed.

mebasoglu, 2023-08-17:
On the `get_pointcloud()` method in `vlp16_decoder` I changed the angle type from radians to degrees like the previous `awf_velodyne` driver and it worked.

Instead of these:
https://github.com/tier4/nebula/blob/f82a6f86748a3c86a703bcc229ca2f5eddb933c2/nebula_decoders/src/nebula_decoders_velodyne/decoders/vlp16_decoder.cpp#L63-L64

I added:
```C++
auto current_azimuth_deg = angles::to_degrees(scan_pc_->points.back().azimuth);
auto phase_diff_deg =
      static_cast<size_t>(360 + current_azimuth_deg - sensor_configuration_->scan_phase) % 360;
```

And the new while loop:
```C++
while (phase_diff_deg <= 180 && !scan_pc_->points.empty()) {
      overflow_pc_->points.push_back(scan_pc_->points.back());
      scan_pc_->points.pop_back();
      
      // current_azimuth = scan_pc_->points.back().azimuth;
      // phase_diff = static_cast<size_t>(angles::to_degrees(2*M_PI + current_azimuth - phase)) % 360;

      current_azimuth_deg = angles::to_degrees(scan_pc_->points.back().azimuth);
      phase_diff_deg =
        static_cast<size_t>(360 + current_azimuth_deg - sensor_configuration_->scan_phase) % 360;
    }
```

And now the gap is gone. I have only tested for VLP-16.

amc-nu, 2023-08-21:
Fixed in https://github.com/tier4/nebula/pull/40
and
https://github.com/tier4/nebula/pull/50

mebasoglu, 2023-08-23:
Hello, I still face this issue with most recent commits on main branch.

[image]

drwnz, 2023-08-24:
@mebasoglu thank you, do you have a rosbag with packets topic you can share?

mebasoglu, 2023-08-25:
Hello @drwnz , while working on the vehicle I realized that it was my mistake to set scan phase properly. After configuring it correctly, there is no problem. I apologize for taking your time. Thank you.

closed 2023-08-25 by mebasoglu