tier4/nebula #60: Enhancement: Add Ring-Based Filter for Removing Reflected and Vehicle Points

mebasoglu, 2023-08-29:
## Overview
The proposed enhancement introduces a ring-based filtering mechanism to the existing functionality of the repository. This feature aims to remove unwanted points from specified rings within defined angle ranges, specifically targeting reflected and vehicle points directly within the driver. The intention is to elude crop box filters to decrease the delay of the sensing pipeline.

## Details
The key aspects of this feature include:

1. **Point Removal:** The feature allows for the removal of points based on their position within specified rings.
  
2. **Ring and Angle Specification:** Users will be able to define which rings they want to apply filtering to. Each ring will be associated with one or more angle ranges for removal.
  
3. **Angle Ranges:** For each specified ring, users can define multiple angle ranges within which points should be removed. These angle ranges are determined by the starting and ending angles.
  
4. **Fine-Tuned Filtering:** The ability to define multiple angle ranges for a single ring provides fine-tuned control over the filtering process, enabling users to selectively remove points.
  

**Usage Example:**

```xml
<arg name="invalid_point_remove" default="true"/> <arg name="invalid_rings" default="[3, 5, 7, 7]"/> <arg name="invalid_angles_start" default="[120, 250, 180, 230]"/> <arg name="invalid_angles_end" default="[200, 300, 220, 245]"/>
```

**Expected Behavior:** Upon enabling the `invalid_point_remove` option and specifying the relevant `invalid_rings`, `invalid_angles_start`, and `invalid_angles_end` configurations, the ring-based filter will remove points located within the defined angle ranges for the specified rings.

Could you please provide any feedback, suggestions, and guidance on the proposed enhancement @drwnz @amc-nu ?

mebasoglu, 2023-08-29:
Yes, I thought a decoder level implementation for each sensor, so that we can deactiavte crop box filters to remove vehicle points. 
It could be also useful to remove reflected points from the vehicle's glasses:

[image]

mojomex, 2025-02-17:
This concept has been superseded by the more general #250 mask-based filter. We should be able to make it available for Velodyne pretty quickly, c.f. #251.