tier4/nebula #83: Why duplicating `velodyne_msgs`?

VRichardJP, 2023-10-13:
Messages defined in nebula's `velodyne_msgs` package are already defined in the widely used `velodyne_msgs` package:
https://github.com/ros-drivers/velodyne/tree/master/velodyne_msgs

If the point is to be able to modify the messages in the future, it would be better to rename the package to avoid name collision, for example `nebula_velodyne_msgs`.
If the point is to use the exact same definition than the original `velodyne_msgs`, then it makes more sense to add an external dependency to `velodyne_msgs` instead of duplicating it.

No?

drwnz, 2023-10-20:
Thank you for the comment.
Actually, I plan to remove both `velodyne_msgs` and `pandar_msgs` and create `nebula_msgs` so that we can unify all sensors to a single packet message type. 
In the meantime, either of your suggestions makes sense - probably we can add the dependency.
@amc-nu do you have any additional comments on this?

xmfcx, 2023-11-15:
Related issue:
- https://github.com/autowarefoundation/autoware/issues/3961#issuecomment-1812653045

drwnz, 2023-11-16:
@xmfcx thank you for linking this - it seems it is causing confusion, so I will remove for now, add the `velodyne_msgs` dependency and we will make an announcement when `nebula_msgs` is complete for LiDAR packet messages.

drwnz, 2023-11-16:
Fixed with https://github.com/tier4/nebula/pull/93

closed 2023-11-16 by drwnz