tier4/nebula #89: No calibration files required

JianWangInno, 2023-11-07:
For Innovusion's LiDAR, data calibration is all done within the LiDAR itself, and no calibration is required on the client side. However, in the existing framework, the system cannot run properly without providing a calibration file. Can the launch script be modified to remove this restriction?

amc-nu, 2023-11-13:
The project currently does not have decoders for Innovusion's Lidar. However, if implemented from scratch, it should be possible to create an Innovusion HW connection and Packet Decoder inheriting from NebulaHwInterfaceBase and NebulaDriverBase, respectively.

JianWangInno, 2023-11-13:
> The project currently does not have decoders for Innovusion's Lidar. However, if implemented from scratch, it should be possible to create an Innovusion HW connection and Packet Decoder inheriting from NebulaHwInterfaceBase and NebulaDriverBase, respectively.

Thank you for your reply. Currently, Lidar from innovusion has submitted a PR and will face this issue once merged.

amc-nu, 2023-11-13:
Thank you for providing additional information, and thank you for your [PR](https://github.com/tier4/nebula/pull/90).
In this case, creating a new launch file might be better. Starting from ROS2 Humble, we recommend XML files.
The following link shows a tutorial on using composable nodes within XML launch files.
https://docs.ros.org/en/humble/How-To-Guides/Launching-composable-nodes.html

JianWangInno, 2023-11-13:
> Thank you for providing additional information, and thank you for your [PR](https://github.com/tier4/nebula/pull/90). In this case, creating a new launch file might be better. Starting from ROS2 [...]

Thank you for your solution. We will support it in the next version.

JianWangInno, 2023-11-14:
> Thank you for providing additional information, and thank you for your [PR](https://github.com/tier4/nebula/pull/90). In this case, creating a new launch file might be better. Starting from ROS2 [...]

> In ROS2, XML Launch files do not support defining composite nodes in the same way as ComposableNode in Python. XML Launch files are primarily used for starting and configuring nodes and do not provide the same level of advanced programming capabilities as Python code.

When we use XML launch files to start Nebula, we found that the hw_interface, monitor, and decoder nodes are running in separate processes, which can lead to performance degradation. On the other hand, using Python launch files allows us to utilize the concept of composite nodes, where all three nodes run in the same process. Based on my research, XML launch files currently do not support composite nodes. To ensure efficiency, would it be possible for us to modify Nebula's nebula_launch.py? Alternatively, could we create a new file called nebula_launch_innovusion.py to address this issue with multiple processes?

amc-nu, 2023-11-15:
You can create a component container and add the composable nodes. Please look at the link I shared above. It contains an example. Click on the **XML** tab.

https://docs.ros.org/en/humble/How-To-Guides/Launching-composable-nodes.html#launch-file-examples

JianWangInno, 2023-11-15:
I have tried according to the document you provided, but unfortunately, the error message is as follows. It should be a lack of third-party packages or frameworks to support some of the XML tags. Due to time constraints, we have shelved the XML integration combination node solution and continued to use the PY method to start it. We have added a specific startup command line in README (sensor_ip must be configured). Thank you for your continuous attention.

`launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [xml]: Unrecognized entity of the type: node_container`
`launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [xml]: Unrecognized entity of the type: load_composable_node`

amc-nu, 2023-11-15:
Can you please paste the contents of your XML file?

JianWangInno, 2023-11-15:
FYI 

```
<launch>
  <arg name="sensor_model" description="Robin/Falcon"/>
  <arg name="return_mode" default="Dual" description="See readme for supported return modes"/>
  <arg name="frame_id"  default="innovusion"/>
  <arg name="sensor_ip" default="172.168.1.10" description="Lidar Sensor IP"/>
  <arg name="host_ip" default="172.168.1.11" description="Local IP"/>
  <arg name="data_port" default="8010" description="LiDAR Data Port"/>
  <arg name="cloud_min_range" default="0.4" description="Min Point Range"/>
  <arg name="cloud_max_range" default="2000.0" description="Max Point Range"/>
  <arg name="setup_sensor" default="true" description="Enable sensor setup on hw-driver."/>

    <node_container pkg="rclcpp_components" exec="component_container" name="innovusion" namespace="">
        <composable_node pkg="nebula_ros" exec="innovusion_driver_ros_wrapper_node" plugin="innovusion_driver_ros_wrapper_node" name="innovusion_cloud" namespace="">
            <param name="sensor_model" value="$(var sensor_model)"/>
            <param name="host_ip" value="$(var host_ip)"/>
            <param name="return_mode" value="$(var return_mode)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="sensor_ip" value="$(var sensor_ip)"/>
            <param name="cloud_min_range" value="$(var cloud_min_range)"/>
            <param name="cloud_max_range" value="$(var cloud_max_range)"/>
        </composable_node>
        <composable_node pkg="nebula_ros" exec="innovusion_hw_interface_ros_wrapper_node" plugin="innovusion_hw_interface_ros_wrapper_node" name="innovusion_hw_driver" namespace="">
            <param name="sensor_model" value="$(var sensor_model)"/>
            <param name="return_mode" value="$(var return_mode)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="sensor_ip" value="$(var sensor_ip)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="host_ip" value="$(var host_ip)"/>
            <param name="data_port" value="$(var data_port)"/>
            <param name="setup_sensor" value="$(var setup_sensor)"/>
            <param name="cloud_min_range" value="$(var cloud_min_range)"/>
            <param name="cloud_max_range" value="$(var cloud_max_range)"/>
        </composable_node>
        <composable_node pkg="nebula_ros" exec="innovusion_hw_monitor_ros_wrapper_node" plugin="innovusion_hw_monitor_ros_wrapper_node" name="innovusion_hw_monitor" namespace="">
            <param name="sensor_model" value="$(var sensor_model)"/>
            <param name="return_mode" value="$(var return_mode)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="sensor_ip" value="$(var sensor_ip)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="host_ip" value="$(var host_ip)"/>
            <param name="data_port" value="$(var data_port)"/>
            <param name="setup_sensor" value="$(var setup_sensor)"/>
            <param name="cloud_min_range" value="$(var cloud_min_range)"/>
            <param name="cloud_max_range" value="$(var cloud_max_range)"/>
        </composable_node>
    </node_container>
</launch>
```

amc-nu, 2023-11-15:
Thank you
Can I test this for with your branch?

JianWangInno, 2023-11-15:
Of course you can. If you want to test whether the AA tag in the XML startup file is available, we don't need our device to verify it. You can start it as follows:

`ros2 launch nebula_ros my_launch.xml sensor_model:=Falcon`

my_launch.xml contains the XML code posted above.

amc-nu, 2023-11-16:
I found two main issues:
1. Since you want to use a composable node, you need to use `plugin`, not `exec`. Your file contains both, so it fails, the same would happen on a Python launch file.
2. The plugin name needs to be corrected. Please use the name in your CMakeLists.txt.

I didn't have a sensor, but I could launch successfully with the following file
```
<launch>
    <arg name="sensor_model" description="Robin/Falcon"/>
    <arg name="return_mode" default="Dual" description="See readme for supported return modes"/>
    <arg name="frame_id"  default="innovusion"/>
    <arg name="sensor_ip" default="172.168.1.10" description="Lidar Sensor IP"/>
    <arg name="host_ip" default="172.168.1.11" description="Local IP"/>
    <arg name="data_port" default="8010" description="LiDAR Data Port"/>
    <arg name="cloud_min_range" default="0.4" description="Min Point Range"/>
    <arg name="cloud_max_range" default="2000.0" description="Max Point Range"/>
    <arg name="setup_sensor" default="true" description="Enable sensor setup on hw-driver."/>

    <node_container pkg="rclcpp_components" exec="component_container" name="innovusion" namespace="">
        <composable_node pkg="nebula_ros" plugin="InnovusionDriverRosWrapper" name="innovusion_cloud" namespace="">
            <param name="sensor_model" value="$(var sensor_model)"/>
            <param name="host_ip" value="$(var host_ip)"/>
            <param name="return_mode" value="$(var return_mode)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="sensor_ip" value="$(var sensor_ip)"/>
            <param name="cloud_min_range" value="$(var cloud_min_range)"/>
            <param name="cloud_max_range" value="$(var cloud_max_range)"/>
        </composable_node>
        <composable_node pkg="nebula_ros" plugin="InnovusionHwInterfaceRosWrapper" name="innovusion_hw_driver" namespace="">
            <param name="sensor_model" value="$(var sensor_model)"/>
            <param name="return_mode" value="$(var return_mode)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="sensor_ip" value="$(var sensor_ip)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="host_ip" value="$(var host_ip)"/>
            <param name="data_port" value="$(var data_port)"/>
            <param name="setup_sensor" value="$(var setup_sensor)"/>
            <param name="cloud_min_range" value="$(var cloud_min_range)"/>
            <param name="cloud_max_range" value="$(var cloud_max_range)"/>
        </composable_node>
        <composable_node pkg="nebula_ros" plugin="InnovusionHwMonitorRosWrapper" name="innovusion_hw_monitor" namespace="">
            <param name="sensor_model" value="$(var sensor_model)"/>
            <param name="return_mode" value="$(var return_mode)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="sensor_ip" value="$(var sensor_ip)"/>
            <param name="frame_id" value="$(var frame_id)"/>
            <param name="host_ip" value="$(var host_ip)"/>
            <param name="data_port" value="$(var data_port)"/>
            <param name="setup_sensor" value="$(var setup_sensor)"/>
            <param name="cloud_min_range" value="$(var cloud_min_range)"/>
            <param name="cloud_max_range" value="$(var cloud_max_range)"/>
        </composable_node>
    </node_container>
</launch>
```

JianWangInno, 2023-11-17:
```
Since you want to use a composable node, you need to use plugin, not exec. Your file contains both, so it fails, the same would happen on a Python launch file.
The plugin name needs to be corrected. Please use the name in your CMakeLists.txt.
```
Firstly, thank you for pointing out the error in the XML startup file. 
However, I still reported the following error when using the XML startup file you provided.
Is it because I did not install the `ros-humble-image-tools` plugin that supports `node_container `nodes?  Is this ROS2 officially supported?

`launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [xml]: Unrecognized entity of the type: node_container`

JianWangInno, 2023-11-17:
FYI. The complete error report is as follows:

```
[INFO] [launch]: All log files can be found below /home/demo/.ros/log/2023-11-17-16-12-01-719210-demo-VirtualBox-38808
[INFO] [launch]: Default logging verbosity is set to INFO
Task exception was never retrieved
future: <Task finished name='Task-2' coro=<LaunchService._process_one_event() done, defined at /opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py:226> exception=InvalidLaunchFileError('xml')>
Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 53, in get_launch_description_from_any_launch_file
    return loader(launch_file_path)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/frontend_launch_file_utilities.py", line 35, in get_launch_description_from_frontend_launch_file
    return parser.parse_description(root_entity)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 111, in parse_description
    actions = [self.parse_action(child) for child in entity.children]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 111, in <listcomp>
    actions = [self.parse_action(child) for child in entity.children]
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/parser.py", line 88, in parse_action
    return instantiate_action(entity, self)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/frontend/expose.py", line 38, in instantiate_action
    raise RuntimeError('Unrecognized entity of the type: {}'.format(entity.type_name))
RuntimeError: Unrecognized entity of the type: node_container

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 228, in _process_one_event
    await self.__process_event(next_event)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_service.py", line 248, in __process_event
    visit_all_entities_and_collect_futures(entity, self.__context))
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 45, in visit_all_entities_and_collect_futures
    futures_to_return += visit_all_entities_and_collect_futures(sub_entity, context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/utilities/visit_all_entities_and_collect_futures_impl.py", line 38, in visit_all_entities_and_collect_futures
    sub_entities = entity.visit(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/action.py", line 108, in visit
    return self.execute(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/actions/include_launch_description.py", line 130, in execute
    launch_description = self.__launch_description_source.get_launch_description(context)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_source.py", line 84, in get_launch_description
    self._get_launch_description(self.__expanded_location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_description_source.py", line 53, in _get_launch_description
    return get_launch_description_from_any_launch_file(location)
  File "/opt/ros/galactic/lib/python3.8/site-packages/launch/launch_description_sources/any_launch_file_utilities.py", line 56, in get_launch_description_from_any_launch_file
    raise InvalidLaunchFileError(extension, likely_errors=exceptions)
launch.invalid_launch_file_error.InvalidLaunchFileError: Caught exception when trying to load file of format [xml]: Unrecognized entity of the type: node_container
```

amc-nu, 2023-11-17:
I see, like I mentioned  [before](https://github.com/tier4/nebula/issues/89#issuecomment-1807667603) this is only for ROS2 Humble.
I see you’re running an outdated ROS2 version, Galactic, which is EOL. We recommend upgrading to Humble.

JianWangInno, 2023-11-17:
I understand that the fundamental issue lies in the version differences of Ros2.

If necessary, we will upgrade the Ros2 version. Thanks.

JianWangInno, 2023-11-20:
I have switched to the Ros2 Humble version and verified the XML file you provided, which proves to be feasible. Thank you.

closed 2023-11-20 by JianWangInno