tier4/nebula #91: UdpDriver does not support receiving multicast data

JianWangInno, 2023-11-13:
The UdpDriver function provided by the Nebula project has a relatively simple interface and does not support joining multicast groups, so multicast data cannot be received now. Do you have any plans to support this feature？

knzo25, 2023-12-15:
@JianWangInno 
The transport drivers are outside the scope of Nebula (for now), so support for that in their main branch should be requested in the corresponding repository.

That being said, I had to use multicast for one of our new sensors, so I just modified the transport drivers to give multicast support
https://github.com/knzo25/transport_drivers/tree/feat/multicast_support

JianWangInno, 2023-12-26:
Thanks for your commit.  I have verified that the multicast mode is indeed feasible. 
[https://github.com/knzo25/transport_drivers/tree/feat/multicast_support] 

So will your changes be submitted as PR to MapIV, or will you directly modify the repository that Nebula depends on to multicast_ Support?

JianWangInno, 2023-12-26:
> Thanks for your [commit](https://github.com/knzo25/transport_drivers/tree/feat/multicast_support). I have verified that the multicast mode is indeed feasible.
> 
> So will your changes be submitted as PR to MapIV, or will you directly modify the repository that Nebula depends on to multicast_ Support?

Based on your submission, my adaptation code is still local and has not been submitted.

knzo25, 2023-12-27:
@JianWangInno 
The multicast was added while developing support for the Continental ARS548.
Right now we are planning some related experiments to evaluate both the sensor and the driver, then we need to discuss with internal stakeholders, and then we will decide on how to proceed with the transport drivers (including the multicast support).

Since it will take some months, I would recommend you use my fork in the meantime. Is there a particular sensor/use case that you need a more expedite handling?

JianWangInno, 2023-12-27:
@knzo25 Thanks for your work. 
Currently, the approach we have taken in the submitted PR is as follows: if the sensor uses multicast, the Nebula driver will convert it to unicast to ensure that the driver can reliably receive point cloud data. Both functionality and performance are unaffected. 

If your requirement for the sensor includes the need for multicast support, then this would rely on your fork. If this isn't an urgent requirement, you can proceed with your regular schedule.

closed 2023-12-27 by JianWangInno