tier4/nebula #122: Improving nebula std::cout logs

ahmeddesokyebrahim, 2024-02-26:
### Description
This issue is open out of the running activities in [Autoware issue for reducing excessive log messages](https://github.com/autowarefoundation/autoware.universe/issues/5539).

As Autoware uses nebula, we are creating this issue to handle some logs coming from nebula while using [Rosbag replay simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/).

While using [Rosbag replay simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/), there are some logging messages coming from nebula classes that are not ROS nodes so the log uses `std::cout`. 
[This](https://github.com/tier4/nebula/blob/76526caf873028c90f64179f4aea0d756a9572f4/nebula_common/src/velodyne/velodyne_calibration_decoder.cpp#L166) and [this](https://github.com/tier4/nebula/blob/76526caf873028c90f64179f4aea0d756a9572f4/nebula_decoders/src/nebula_decoders_velodyne/velodyne_driver.cpp#L17) are examples of source code for the intended logs.

The output in terminal looks very noisy specially in cases laser ring number is bigger ... 
- example 1 
```
[component_container-33] laser_ring[0] =0, angle = -0.261799
[component_container-33] laser_ring[2] =1, angle = -0.226893
[component_container-33] laser_ring[4] =2, angle = -0.191986
[component_container-33] laser_ring[6] =3, angle = -0.15708
[component_container-33] laser_ring[8] =4, angle = -0.122173
[component_container-33] laser_ring[10] =5, angle = -0.0872665
[component_container-33] laser_ring[12] =6, angle = -0.0523599
[component_container-33] laser_ring[14] =7, angle = -0.0174533
[component_container-33] laser_ring[1] =8, angle = 0.0174533
[component_container-33] laser_ring[3] =9, angle = 0.0523599
[component_container-33] laser_ring[5] =10, angle = 0.0872665
[component_container-33] laser_ring[7] =11, angle = 0.122173
[component_container-33] laser_ring[9] =12, angle = 0.15708
[component_container-33] laser_ring[11] =13, angle = 0.191986
[component_container-33] laser_ring[13] =14, angle = 0.226893
[component_container-33] laser_ring[15] =15, angle = 0.261799

```
```
[component_container-33] sensor_configuration->sensor_model=VLP16
```
- example 2
```
[component_container_mt-30] laser_ring[36] =0, angle = -0.436332
[component_container_mt-30] laser_ring[69] =1, angle = -0.34177
[component_container_mt-30] laser_ring[54] =2, angle = -0.279986
[component_container_mt-30] laser_ring[87] =3, angle = -0.236754
[component_container_mt-30] laser_ring[0] =4, angle = -0.204937
[component_container_mt-30] laser_ring[97] =5, angle = -0.180572
[component_container_mt-30] laser_ring[18] =6, angle = -0.161338
[component_container_mt-30] laser_ring[115] =7, angle = -0.14577
[component_container_mt-30] laser_ring[44] =8, angle = -0.133518
[component_container_mt-30] laser_ring[77] =9, angle = -0.124791
[component_container_mt-30] laser_ring[62] =10, angle = -0.119555
[component_container_mt-30] laser_ring[95] =11, angle = -0.116064
[component_container_mt-30] laser_ring[8] =12, angle = -0.113446
[component_container_mt-30] laser_ring[105] =13, angle = -0.111527
[component_container_mt-30] laser_ring[26] =14, angle = -0.109607
[component_container_mt-30] laser_ring[123] =15, angle = -0.107687
[component_container_mt-30] laser_ring[100] =16, angle = -0.105767
[component_container_mt-30] laser_ring[21] =17, angle = -0.103847
[component_container_mt-30] laser_ring[118] =18, angle = -0.101927
[component_container_mt-30] laser_ring[39] =19, angle = -0.100007
[component_container_mt-30] laser_ring[64] =20, angle = -0.0980875
[component_container_mt-30] laser_ring[49] =21, angle = -0.0961676
[component_container_mt-30] laser_ring[82] =22, angle = -0.0942478
[component_container_mt-30] laser_ring[3] =23, angle = -0.0923279
[component_container_mt-30] laser_ring[108] =24, angle = -0.0904081
[component_container_mt-30] laser_ring[29] =25, angle = -0.0884882
[component_container_mt-30] laser_ring[126] =26, angle = -0.0865683
[component_container_mt-30] laser_ring[47] =27, angle = -0.0846485
[component_container_mt-30] laser_ring[72] =28, angle = -0.0827286
[component_container_mt-30] laser_ring[57] =29, angle = -0.0808087
[component_container_mt-30] laser_ring[90] =30, angle = -0.0788889
[component_container_mt-30] laser_ring[11] =31, angle = -0.076969
[component_container_mt-30] laser_ring[52] =32, angle = -0.0750492
[component_container_mt-30] laser_ring[85] =33, angle = -0.0731293
[component_container_mt-30] laser_ring[6] =34, angle = -0.0712094
[component_container_mt-30] laser_ring[103] =35, angle = -0.0692896
[... 89 lines cut ...]
[component_container_mt-30] laser_ring[41] =125, angle = 0.169297
[component_container_mt-30] laser_ring[74] =126, angle = 0.205076
[component_container_mt-30] laser_ring[59] =127, angle = 0.261799

```

```
[component_container_mt-30] sensor_configuration->sensor_model=VLS128
```

These log messages are printed by default and can't be configured to a level to be visible while debug only for example.

The intention of this issue is to feature of configuring the level of log messages that are coming from nebula classes that do not inherit from rclcpp::Node so that we can have these messages as ERROR, WARN, INFO, and DEBUG similar to ROS2 log levels. 

### Expected behavior

Being able to configure level of nebula log messages

### Actual behavior

Log level configuration does not exist for messages that are coming from classes that do not inherit from rclcpp::Node

### Steps to reproduce
1. Follow the steps for [Rosbag replay simulation](https://autowarefoundation.github.io/autoware-documentation/main/tutorials/ad-hoc-simulation/rosbag-replay-simulation/)
2. You can find the intended logs in terminal as explained in the description

mitsudome-r, 2024-02-26:
@drwnz Could you answer the following questions?
 - Are there reasons for not using ROS loggers for [some of stdout messages](https://github.com/tier4/nebula/blob/76526caf873028c90f64179f4aea0d756a9572f4/nebula_common/src/velodyne/velodyne_calibration_decoder.cpp#L166)? (e.g., you don't want to have ROS dependency for these files)
 - At least for the above example, it seems the code expects `calibration.ros_info` to be used as a parameter to configure the output information. However, there seems to be no ROS Parameter (or any other means) to modify this parameter at runtime. Is this intentional?

drwnz, 2024-02-26:
@amc-nu I believe this is just a leftover from the original Velodyne implementation? We can safely remove the stdout output, right?

amc-nu, 2024-02-28:
Several internal libraries are ROS-independent, so these instead use std::cout.
Some of the output is intended as info, so these can be safely removed.

ahmeddesokyebrahim, 2024-02-28:
Thanks @drwnz & @amc-nu for your replies.
I have cleaned these leftover `std::cout` logs as you recommended and included this implementation in this [PR](https://github.com/tier4/nebula/pull/118).
I will mark it as ready for review very soon once I do not find any further changes in that task relevant to nebula

cc: @mitsudome-r @xmfcx

ahmeddesokyebrahim, 2024-03-29:
Closing this issue after merging PR https://github.com/tier4/nebula/pull/118

closed 2024-03-29 by ahmeddesokyebrahim