tier4/nebula #157: Bug in EIGEN_ALIGN16

veqcc, 2024-06-01:
There is a known bug in `EIGEN_ALIGN16` in PCL: https://github.com/leggedrobotics/tensorflow-cpp/blob/master/ISSUES.md

Due to the bug, the clang-tidy make the following CRITICAL warning 
```
expected ';' after struct
```
```
'alignas' attribute only applies to variables, data members and tag types
```
in https://github.com/tier4/nebula/blob/main/nebula_common/include/nebula_common/point_types.hpp
```
struct PointXYZIR
{
  PCL_ADD_POINT4D;
  float intensity;
  uint16_t ring;
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGEN_ALIGN16;
...
struct PointXYZIRADT
{
  PCL_ADD_POINT4D;
  float intensity;
  uint16_t ring;
  float azimuth;
  float distance;
  uint8_t return_type;
  double time_stamp;
  EIGEN_MAKE_ALIGNED_OPERATOR_NEW
} EIGEN_ALIGN16;
```

Could avoid the bug?

mojomex, 2024-06-01:
Hi, thanks for bringing attention to this!

This is a known issue - I would love to use all the Clang tooling for Nebula - but even when fixing this, I found that `transport_drivers` also does not compile with Clang so I gave up for the time being. I will look into this some time in the near future. If you want a fix now, please feel free to PR :bowing_man:

veqcc, 2024-06-02:
@mojomex 
Thank you for your quick reply!!

My first concern is whether the struct `PointXYZIR` and `PointXYZIRADT` are aligned, in spite of the pcl bug?

If they are NOT aligned, it should be solved soon.

If they are aligned, then I think we have two options:
1. use `pcl>=v1.10.0`
It is reccomnended here: https://github.com/leggedrobotics/tensorflow-cpp/blob/master/ISSUES.md
2. do nothing for now, and look into in some future as you say

veqcc, 2024-06-07:
May I close it?

mojomex, 2024-06-07:
@veqcc The fix is currently merged into develop, we'll close the issue once it is merged into main!

mojomex, 2024-07-31:
Fixed by 
* #166

PR #160 (2024-06-05): fix: make Nebula Clang-compatible
## PR Type

- Improvement
- Bug Fix

## Related Links

* Fixes #157 
* For reference: [the same fix in PCL](https://github.com/PointCloudLibrary/pcl/pull/3237)

## Description

This PR moves the `EIGEN_ALIGN16` from behind point structs to after the `struct` keyword, remaining compatible with GCC while enabling compilation with Clang (and enabling usage of all the great Clang tools :tada:)

:warning: Note: Nebula still does not compile on the newest Clang version (18) due to a tricky error in `transport_drivers`. The newest version I have tested with is Clang  [... cut]
closed 2024-07-31 by mojomex