tier4/nebula #179: Request: change to not download calibration file when it is already downloaded

h-ohta, 2024-07-30:
Now, the nebula driver always downloads calibration file from LiDAR sensor when launching driver.
Could you change not to download calibration file when it is already downloaded in specific directory ?

drwnz, 2024-07-31:
I would like to propose that we save the sensor MAC address (either in the file name, or better the first line of the file) and then verify that the sensor is correct before deciding whether to download from the sensor or not.

h-ohta, 2024-07-31:
Thank you.
> were there any particular problems with nebula downloading the data every time?

I have no problem about driver. But, I have long console log when I launch nebula driver.
I'd like to reduce console log as less as possible.