tier4/nebula #197: Missing `boost_tcp_driver` Dependency in `nebula_hw_interfaces` Package

cavadibrahimli1, 2024-09-17:
**Description**:
While trying to build the `nebula_hw_interfaces` package in the **Nebula** repository, I encountered a missing dependency error for the `boost_tcp_driver`. Specifically, the build process fails due to the following missing file:

```bash
/home/username/new_ros2_ws/src/nebula/nebula_hw_interfaces/include/nebula_hw_interfaces/nebula_hw_interfaces_hesai/hesai_hw_interface.hpp:13:10: fatal error: boost_tcp_driver/http_client_driver.hpp: No such file or directory
   13 | #include "boost_tcp_driver/http_client_driver.hpp"
      |          ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
compilation terminated.
```

### Steps I Tried:

1. **Installed all dependencies using `rosdep`**:
   ```bash
   rosdep install --from-paths src --ignore-src -r -y

However, rosdep was unable to resolve the dependency for boost_tcp_driver.

2. Cloned the transport_drivers repository: I followed the suggestion to include transport_drivers (https://github.com/ros-drivers/transport_drivers), which provides udp_driver and serial_driver. However, the boost_tcp_driver file is still missing.

3. Explored Tier IV repositories: I looked into several Tier IV-related repositories but could not locate a package that includes the missing boost_tcp_driver.

### Steps to Reproduce:

1. Clone the Nebula repository into a ROS 2 workspace.

2. Attempt to build the workspace:
   ```bash
   colcon build
   ```
   The build will fail with the above-mentioned error.

## Additional Information:
- ROS Version: ROS 2 Humble
- Operating System: Ubuntu 22.04
- Nebula Branch: Main (latest as of the issue date)

## Expected Behavior:
The nebula_hw_interfaces package should either have the boost_tcp_driver dependency properly defined or include instructions for obtaining it.

mojomex, 2024-09-17:
Hi! I see, that note about transport_drivers in our readme is a bit misleading.
Following the instructions a few lines further down, especially `vcs import src < src/build_depends.repos` will install the correct version of transport_drivers. 
Please let us know if that worked for you 🙏

mojomex, 2024-09-24:
Hi @cavadibrahimli1!
I hope you had a chance to give this one more try. If there were any other problems, feel free to let us know!
Otherwise I'll be closing this issue in a couple of days to keep our backlog manageable :bow:

cavadibrahimli1, 2024-09-24:
Hi @mojomex ,

Thank you for following up! I appreciate your support. I will take a closer look at the solution and try again. If any further issues arise, I’ll make sure to reach out.

Thanks again for your help!

Best regards,
Javad Ibrahimlu

drwnz, 2024-11-25:
@cavadibrahimli1 did this resolve your issue? If so, I will close. Thanks.

cavadibrahimli1, 2024-11-25:
Yeah, it solved @drwnz

closed 2024-11-25 by mojomex