tier4/nebula #254: Build Error in nebula_ros Due to Incorrect Use of std::stringstream

Sanchitkedia, 2025-01-27:
System Information:

    Operating System: Ubuntu 22.04
    ROS Version: ROS 2 Humble
    GCC Version: GCC 10.5.0
    G++ Version: G++ 10.5.0
    Build Tool: Colcon

Observed Behavior:

The build fails with errors related to the use of std::stringstream. The error is caused by attempting to call .str() on an std::ostream object, which does not support this function.

Compilation Error:
  `error: ‘std::ostream’ {aka ‘class std::basic_ostream<char>’} has no member named ‘str’`

List of Affected Files:
  - src/continental/continental_srr520_hw_interface_wrapper.cpp
  - src/continental/continental_srr520_decoder_wrapper.cpp
  - src/continental/continental_ars548_hw_interface_wrapper.cpp
  - src/continental/continental_ars548_decoder_wrapper.cpp
  - src/continental/continental_srr520_ros_wrapper.cpp
  - src/continental/continental_ars548_ros_wrapper.cpp
  - src/robosense/hw_interface_wrapper.cpp
  - src/robosense/robosense_ros_wrapper.cpp
  - src/velodyne/hw_interface_wrapper.cpp
  - src/velodyne/velodyne_ros_wrapper.cpp
  - src/velodyne/decoder_wrapper.cpp
  - src/hesai/hw_interface_wrapper.cpp
  - src/hesai/hesai_ros_wrapper.cpp
  - src/hesai/decoder_wrapper.cpp

Example of the Problematic Code
  `File: src/continental/continental_srr520_hw_interface_wrapper.cpp`
  `std::stringstream{} << "Could not initialize HW interface: " << status_).str();`

mojomex, 2025-01-28:
Thank you for opening the issue! You're absolutely right and we have actually fixed this in some places (we have a [helper function](https://github.com/tier4/nebula/blob/33e77cfa081b7beb648aa929cdb70451e5f95d66/nebula_common/include/nebula_common/util/string_conversions.hpp#L38-L44) that does it correctly), but it seems we've missed quite some places.
Will address!

mojomex, 2025-01-28:
@Sanchitkedia Could you tell us the version of Nebula you are using?

closed 2025-01-31 by mojomex

mojomex, 2025-01-31:
Fixed, thanks for reporting! It will be a little while until the next official release, so please look into using main as-is or cherry-picking the changes into v0.2.1 for now 🙏

commit 37724a0b (2025-01-31): chore(nebula_ros): replace non-standard stringstream usage with nebula::util::to_string

Fixes #254.