tier4/nebula #262: Apply distance corrections to obtain distance from geometric center (Hesai)

shrijitsingh99, 2025-02-06:
Hesai LiDAR's report the distance from the optical center and need to apply a correction to get the distance from the geometrical center
https://github.com/HesaiTechnology/HesaiLidar_SDK_2.0/blob/5c5cf656523cbd8426d165549fbdc21e0d219f39/libhesai/UdpParser/src/general_parser.cc#L291

For most use cases, people would be using the geometric center as reference for all their TF's, so would be good to have an option to apply this correction.

mojomex, 2025-02-07:
Hi, thanks for the issue.
I think you're referring to the yellow vs. red dots in this technical drawing (this one is from the XT32 user manual):

[image]

As the datasheet states, all measurements are relative to the origin -- so the HesaiLidar_SDK is either outdated or works with some nonstandard data stream we don't have access to.

For peace of mind, we confirmed that the distances are accurate with the following setup:

An XT32 placed centered at the `1m` mark of a measurement tape with two plastic boxes at the `0m` and `2m` mark respectively (such that the geometric center is exactly `1m` away from each of the boxes).
If the measurements needed correction, we would expect the `31.5mm` the optical center is apart from the geometric center to be missing on each side. In other words, we would expect the boxes to be at `96.85cm` if correction is needed, and `100.00cm` if it is not needed.

Experiment setup:
[image]

Results in Foxglove (the grid is of side length `200cm`, each subdivision being `1cm`):
[image]

As shown, the boxes are (up to a small error) at `100cm` from the center each.

closed 2025-02-07 by mojomex

mojomex, 2025-02-07:
The one on the Hesai website is rounded to the nearest degree while the one in Nebula is from a real sensor. Both will produce an error when used with a different sensor. The reason we use the calibration of a specific sensor is that we have real-world ground truth data for unit tests that were recorded using a real-world sensor. Thus, the unit tests need to use that real-world calibration file as well.