tier4/nebula #310: Velodyne: No pointclouds in udp_only mode unless host_ip is set to broadcast_ip

mbharatheesha, 2025-04-24:
When using `udp_only=true` for a VLP16 usecase, I notice that pointclouds are only available on `/velodyne_points` only if `host_ip=255.255.255.255`. If it is set to the actual `host_ip`, no pointclouds are published.

However, with a Hesai LiDAR with `udp_only=true` and a specific `host_ip`, pointclouds are published to `pandar_points`. In fact, if `host_ip` is set to a broadcast ip address, the driver errors out as expected.

mojomex, 2025-04-25:
Thanks for the feedback!

That's interesting. The real host IP and `0.0.0.0` should definitely work. `255.255.255.255` should be disallowed, but such checks are not implemented in `transport_drivers`.

This is probably the right time to transition to the new UDP socket implementation that is independent of `transport_drivers` (see #235).

I can do that pretty quickly, but would you be willing to test my PR with a real device? I don't have one to check for the next three weeks.

mbharatheesha, 2025-04-25:
@mojomex I can definitely test this if the fix is implemented.. Thank you for the quick response.

drwnz, 2025-05-22:
@mbharatheesha did this solve your issue?

mbharatheesha, 2025-05-22:
@drwnz Yes, this issue is fixed. Sorry I didn't close it sooner.

closed 2025-05-22 by mbharatheesha