tier4/nebula #311: Multi sensors

thundersd, 2025-04-25:
Hi Authors,

I wonder if there is any way to get multiple sensors running and fuse their data (for example, Hesai and Velodyne)?

Thank you!

knzo25, 2025-04-25:
@thundersd 
I do not think this is something that belongs in the driver.
You could instantiate several nebula nodes and use pcl to concatenate them yourself.
Autoware does this in the pointcloud preprorcessor if you are interested

mojomex, 2025-04-25:
@thundersd 
Also see: https://autowarefoundation.github.io/autoware_universe/main/sensing/autoware_pointcloud_preprocessor/docs/concatenate-data/
You would need to follow the Autoware documentation to get this running in the configuration you need.

Closing this since it is not directly related to Nebula.

closed 2025-04-25 by mojomex