tier4/nebula #319: Integrating Hesai JT16

ARK3r, 2025-05-13:
What are the steps to integrate support for the [Hesai JT16 lidar](https://www.hesaitech.com/product/jt16/)?

newkillerwhale, 2025-05-13:
这是个好问题，我也刚买这个雷达

mojomex, 2025-05-15:
Hi @ARK3r, thank you for your interest in Nebula!

We have some documentation on how we designed the [Hesai decoder](https://tier4.github.io/nebula/hesai_decoder_design/), and a [tutorial branch](https://github.com/tier4/nebula/compare/main...tutorial) detailing most of the changes necessary to integrate a new Hesai-like sensor.

JT16 seems to be similar to other Hesai sensors
* overall working principle
* in terms of decoding and calibration
* packet structure
* functional safety info

but different
* in terms of connection mode (serial (RS485) instead of UDP/IP)
* in terms of multiplexed payloads on the same connection (pointcloud, IMU, fault info)

So, in addition to the links above, it would be necessary to
* implement an RS485 driver (probably simpler than [UDP](https://github.com/tier4/nebula/blob/e4a491fe19694f3b72ec531c3e9997185217a61b/nebula_hw_interfaces/include/nebula_hw_interfaces/nebula_hw_interfaces_common/connections/udp.hpp))
* wrap or extend the [decoder](http://github.com/tier4/nebula/blob/main/nebula_decoders/include/nebula_decoders/nebula_decoders_hesai/decoders/hesai_decoder.hpp) to work with multiplexed payloads (or ignore IMU and functional safety info for now)

For review and testing, PCAPs recorded with a real sensor should be enough for us.

Please let us know if you're interested in contributing, we'd love to support JT16 but don't have the sensor ourselves at the moment.